autonomousvision/carla_garage #1: config.pickle file Empty

techner3, 2023-06-21:
Hi , First of all thanks for the sharing the code and pre-trained weights of TF++.

I am trying to run the TF++ Sensor Agent with pre-trained weights downloaded from the google drive link given. It seems that the config.pickle file is empty in all the folders. The size of the config.pickle file is 0 bytes. Can you help me with this ?

Thanks in advance !

Kait0, 2023-06-21:
Hm, I can't reproduce this. When I download the zip file and unpack it all the config.pickle files are 27 kb large.
Would need some additional infos.
Maybe something went wrong with the download or unzip?
Maybe there is some security on your system that deletes the file (pickle is a very general format that can contain code)?

techner3, 2023-06-21:
Thanks for your reply, Can you let me know what kind of information is inside the pickle file ?

Kait0, 2023-06-21:
It contains the [config.py](https://github.com/autonomousvision/carla_garage/blob/main/team_code/config.py) class.
The python file contains the default configuration. These configurations can be changed using [argparse](https://github.com/autonomousvision/carla_garage/blob/b666bdedb74d4de31d81b4ba96ac64c995faa1f2/team_code/train.py#LL380C7-L380C7) when training a model and the new config instance is stored in [config.pickle](https://github.com/autonomousvision/carla_garage/blob/b666bdedb74d4de31d81b4ba96ac64c995faa1f2/team_code/train.py#L583).
During inference we [load](https://github.com/autonomousvision/carla_garage/blob/b666bdedb74d4de31d81b4ba96ac64c995faa1f2/team_code/sensor_agent.py#LL65C1-L65C1) the pickle file and automatically update all changed parameters in the config instance.

I think this is a quite elegant way for configuration management because it has no external dependencies and ensures backwards compatibility. New parameters in the config will be loaded from the default, existing parameters will be updated to match the model you are executing and delete parameters will get added back. Will add some documentation on how it works eventually.

For your problem if you for some reason can't use the pickle file, all parameters that were changed are also documented as text in args.txt and you could manually change the once that are different in the config.py file and just don't load the config.pickle.

techner3, 2023-06-21:
Thanks, Will try that !

closed 2023-06-23 by Kait0