autonomousvision/carla_garage #3: Questions about reproducing Interfuser and TCP.

SmashChen, 2023-07-07:
Thank you for your great job! 
I noticed in the appendix that you tried to reproduce the results of Interfuser and TCP on the Carla leaderboard. How did you reproduce it? Do you use their pre-trained weights directly or use their code to retrain on your dataset.

Kait0, 2023-07-07:
For Interfuser we used their pre-trained weights, because no dataset was released.
For TCP we downloaded their dataset and retrained the TCP model on all of their data.

SmashChen, 2023-07-07:
What confuses me is why you used the pre-trained weights from Interfuser, but only got half the score?

Kait0, 2023-07-07:
Well, I am confused as well.
We talked to the authors, they said the released model is not the same as the one they reported in their paper.
They were vague on what the difference exactly is but emphasized that the released model was trained with less data.
They did not want to release the model from the paper or help us reproduce their results, so we put the numbers from the released model in the paper.

SmashChen, 2023-07-07:
Well，i get it.  Thanks for your reply!

closed 2023-07-07 by Kait0