autonomousvision/carla_garage #6: Assert Image_seq_len==1

raozhongyu, 2023-07-12:
I notice that the Image_seq_len is assert as 1.  I'd like to konw if the author have try the image_seq_len as other number and what's the performance. Meanwhile, I  also like to konw if I want change the image_seq_len, what should I notice?

Kait0, 2023-07-12:
We have not tried using multiple image frames and it is not implemented (that is why the assert exists). We have tested multiple LiDARs as described [here](https://github.com/autonomousvision/carla_garage/blob/main/docs/additional_features.md#temporal-lidar-inputs).
If you want to change in image_seq_len you need to change data.py, your backbone architectures, and sensor_agent.py accordingly. You can look at how lidar_seq_len works for an example.

raozhongyu, 2023-07-12:
Thanks a lot.

closed 2023-07-12 by raozhongyu

raozhongyu, 2023-09-13:
Hi, I notice that the bev semantic prediction is in range [64, 64] which is same as the lidar resolution. But the paper only use the front image, and the paper says that the semantic prediction is only based on the visible area. I don't find how to  convert the bev semantic label to the   visible area? Could you give me some suggestions.

Kait0, 2023-09-14:
[A mask](https://github.com/autonomousvision/carla_garage/blob/f91aabfcbd37edfc1ab37412e968b59b74e7d0c5/team_code/model.py#L427C37-L427C57) is applied to the regions not visible in the camera.