autonomousvision/carla_garage #7: About docker configuration.

SmashChen, 2023-07-18:
Hi authors, Thank you for your contribution!
I noticed that you are also performing Carla Leaderboard benchmark recently. Congratulations on refreshing your score again. I have a question about docker configuration and would like to ask you:
When I run "make_docker.sh" using the "Dockerfile.master" you provided on T-PAMI 22, it throws an error: ERROR: failed to solve: nvidia/cuda:10.2-cudnn7-devel-ubuntu16.04: docker .io/nvidia/cuda:10.2-cudnn7-devel-ubuntu16.04: not found.
I think it's because the Ubuntu16.04 image has been removed, but I can't find a solution.

Kait0, 2023-07-18:
It looks to me like nvidia recently took down a lot of older cuda versions from their docker hub.
I don't know if it is possible to still download them somehow.
For this repository I [updated the cuda version](https://github.com/autonomousvision/carla_garage/blob/main/tools/Dockerfile.master) to 11.3.1

`FROM nvidia/cuda:11.3.1-cudnn8-devel-ubuntu16.04`

You can check the available versions [here](https://hub.docker.com/r/nvidia/cuda/tags?page=1).

I am using cuda 11.3 for pytorch in this repository as well. You might also need to adjust the pytorch, mmdet cuda versions etc. for things to work out with the transfuser repository.

Kait0, 2023-07-19:
You just need to make a team, apply for the benchmark and then wait.
The team is manually approved by the staff. I don't think there is any way to speed things up if you already asked them.

SmashChen, 2023-07-19:
Ok, i get it. Sincerely thank you for your reply.

closed 2023-07-20 by Kait0