autonomousvision/carla_garage #15: Has NC TF been released?

wheltz, 2023-10-26:
Hello, this is very helpful work. I would like to ask if the published code contains the implementation of NC conditioned model of TF in the paper. I want to do some extension work, thank you very much

Kait0, 2023-10-26:
Yes all experiments in the paper can be run with the released codebase.
The option for conditioning signal is `--use_tp 0` to remove the target point and `--use_discrete_command 1` to use the navigational command.

closed 2023-10-26 by wheltz

Wolfybox, 2023-11-07:
> Yes all experiments in the paper can be run with the released codebase. The option for conditioning signal is `--use_tp 0` to remove the target point and `--use_discrete_command 1` to use the navigational command.

[image]

hello, I found that all provided pretrained models are trained with use_tp=1. Is it ok to test them with option use_tp=0?