autonomousvision/carla_garage #17: Correct way to test NC conditioned model

Wolfybox, 2023-11-07:
I tried running Transfuser++ with use_tp = False and use_discrete_command=True, but the ego-vehicle kept turning right (until it ran into some obstacles) and wouldn't follow the route. Any ideas about this?

Kait0, 2023-11-07:
What exactly are you trying to do?
Changing the inputs only during inference makes not much sense as the model expects the same data as it was trained with.
If you want to do an ablation like in table 1 you first need to retrain the model with that settings.
I haven't run that particular experiment in the final setting, but I would expect similar results to table 1.

Wolfybox, 2023-11-07:
> What exactly are you trying to do? Changing the inputs only during inference makes not much sense as the model expects the same data as it was trained with. If you want to do an ablation like in [...]

Thanks for replies. I just wanna test a NC model on a few routes to validate its capability of driving with only discrete nav commands. Retraining would be computationally expensive. Could you provide pretrained weights for that?

Kait0, 2023-11-07:
https://drive.google.com/file/d/1ijtgKwZuZ7GYnZhf767yAmc0z-nADVa-/view?usp=sharing
These are the NC models from Table 1.
Index 009 corresponds to without augmentation
Index 010 corresponds to with augmentation.

Wolfybox, 2023-11-08:
> https://drive.google.com/file/d/1ijtgKwZuZ7GYnZhf767yAmc0z-nADVa-/view?usp=sharing These are the NC models from Table 1. Index 009 corresponds to without augmentation Index 010 corresponds to with augmentation.

Millions of thanks for the weights ! 
Now there comes a new problem. I try these weights and it gave me the following errors:

``` shell
File "/home/user/Downloads/wolfybox/proj/transfuser/team_code_tfpp/sensor_agent.py", line 584, in run_step
    raise ValueError('An output representation was chosen that was not trained.')
ValueError: An output representation was chosen that was not trained.
```

the code block (sensor_agent.py at line:554) that raise this error:
``` python
if self.config.inference_direct_controller and self.config.use_controller_input_prediction:
      steer, throttle, brake = self.nets[0].control_pid_direct(pred_target_speed, pred_angle, gt_velocity)
    elif self.config.use_wp_gru and not self.config.inference_direct_controller:
      steer, throttle, brake = self.nets[0].control_pid(self.pred_wp, gt_velocity)
    else:
      raise ValueError('An output representation was chosen that was not trained.')
```

some critical config values at runtime:
[image]

Kait0, 2023-11-08:
export DIRECT=0
you need to set the environment valiable to 0 since this model uses the waypoint output.

Wolfybox, 2023-11-08:
> export DIRECT=0 you need to set the environment valiable to 0 since this model uses the waypoint output.

With DIRECT=0, it works now. Thank you a lot.

closed 2023-11-08 by Wolfybox