autonomousvision/carla_garage #18: Confused about implementation of GlobalRoutePlanner

wheltz, 2023-11-11:
During the process of debugging the `interpolate_trajectory` code, I found that there are often some duplicate waypoints (usually corresponding to different `RoadOptions`) in the `interpolated_trace` obtained from `GlobalRoutePlanner`. After reading the implementation of `trace_route` in `GlobalRoutePlanner`, I found that this may be because when routes contain multiple sub-routes, the exit_waypoint of the route will be calculated twice. Is this design intentional and helpful for navigation?

wheltz, 2023-11-13:
I don't know if this question is less important. When I tried to re-implement interpolate_trajectory in map track for carla, I got confused about the implementation

Kait0, 2023-11-13:
I don't know, the code you mentioned comes from the carla leaderboard / carla.
The interpolate trajectory I copied from the [CARLA leaderboard](https://github.com/autonomousvision/carla_garage/blob/8bb143807ee9df6c1524fe72043ea687ad2ec2aa/leaderboard/leaderboard/utils/route_manipulation.py#L132) to reconstruct the same dense waypoints they were using.
Do you think the current behavior causes any problems?

wheltz, 2023-11-13:
Thanks for your reply. I previously tried to use a buffer to store denser waypoints of the local planner between adjacent target points. I found that a bug in my program came from the implementation of `trace_route`, which I felt was slightly unnatural (of course I can modify the implementation of my program to for normal use). But I'm not sure if it is possible to directly modify `global_route_planner` from `scenario_runner`. As you can see, I'm new to autonomous driving tasks and I don't understand many of the rules of carla leaderboard task

Kait0, 2023-11-13:
Well all the code is open source, so you can copy the function that you need to your file and modify them to fit your need.
You shouldn't modify the leaderboard client itself if possible to keep comparable settings with other work, but inside your agent code you can use what code you want (you should make sure it isn't accessing the simulator itself, to keep the setting non-priviledged, but I think with the HD-map sensor you should have all necessary infos to do route planning)

wheltz, 2023-11-14:
Thank you for your response. I feel very fortunate that the autonomous driving open-source community has such enthusiastic and sharing contributors like you.

closed 2023-11-15 by wheltz