autonomousvision/carla_garage #24: Question about the "route" in measurements file

raozhongyu, 2024-01-09:
Hi, Thanks for your great work. I'd like to use the code to save the hd_map data. I notice that the route in measurements seems like to be the map information. Can you give me some details. Thanks a lot?

Kait0, 2024-01-09:
The route as well as the bev segmentation are created using HD-Map information.
We don't save the entire HD-map in the dataset.
If you just want to store the whole HD-map on disk you could just store the string you get from the HD-map sensor in a .txt file and run that code for every map. 
You can see how to make a carla.Map object from the HD-map string [here](https://github.com/autonomousvision/carla_garage/blob/fb4b180332e465c53095ad6de29429442664882c/team_code/map_agent.py#L288C7-L288C7) (e.g. you can load the string later once you need it).

raozhongyu, 2024-01-09:
Thanks a lot. I will try it.

发自我的iPhone

------------------ Original ------------------
From: Bernhard Jaeger ***@***.***&gt;
Date: Tue,Jan 9,2024 10:26 PM
To: autonomousvision/carla_garage ***@***.***&gt;
Cc: raozhongyu ***@***.***&gt;, Author ***@***.***&gt;
Subject: Re: [autonomousvision/carla_garage] Question about the "route" inmeasurements file (Issue #24)

yihang99, 2024-01-09:
Hello, thank you for your great work first. I'd like to ask the following question about your latest paper transfuser++:

In table6 and table7 you mentioned that the longset6 is used as training towns and LAV is used as validation. Does that mean that you collected the expert data (for imitation) in routes of longest6? 

Also, in appendix C I see that 'we test our additions to TransFuser by repeating the experiments on the training towns (Longest6) while training with all data. ' I wonder what does 'all data' mean here, comparing to the dada you used before.

Shortly, I am kind of confused by the category of the data you used (By town or by route?). I would be greatful if you can make it clearer to me. : )

closed 2024-01-10 by raozhongyu

raozhongyu, 2024-01-16:
Hi, I have collect the surrounding hd map information based on the exsiting dataset. I  meet a little problem, where I only want to collect the front information of ego-car. I'd like to konw how to get the yaw and roation  based on the exsiting information in your dataset such as the ego matrix, the theta.

Kait0, 2024-01-17:
https://github.com/autonomousvision/carla_garage/blob/fb4b180332e465c53095ad6de29429442664882c/team_code/autopilot.py#L359 The global yaw of the ego vehicle is stored in the measurement file as theta.