autonomousvision/carla_garage #29: setup_carla.sh does not find the download files

simensaxe, 2024-02-29:
Hi,

Thank you for an amazing repo! I run in to a problem when installing carla with your setup_carla.sh script.  I get the following error:

```
https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.10.1.tar.gz
Resolving carla-releases.s3.eu-west-3.amazonaws.com (carla-releases.s3.eu-west-3.amazonaws.com)... 16.12.18.6, 16.12.20.14
Connecting to carla-releases.s3.eu-west-3.amazonaws.com (carla-releases.s3.eu-west-3.amazonaws.com)|16.12.18.6|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-02-29 08:59:44 ERROR 404: Not Found.

https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/AdditionalMaps_0.9.10.1.tar.gz
Resolving carla-releases.s3.eu-west-3.amazonaws.com (carla-releases.s3.eu-west-3.amazonaws.com)... 16.12.18.6, 16.12.20.14
Connecting to carla-releases.s3.eu-west-3.amazonaws.com (carla-releases.s3.eu-west-3.amazonaws.com)|16.12.18.6|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-02-29 08:59:44 ERROR 404: Not Found
```

Thank you for all help!

Kait0, 2024-02-29:
Looks like the AWS servers for the CARLA release is just down.
Perhaps you can try the dockerhub CARLA instead (just need to run CARLA inside the docker container then).
`docker pull carlasim/carla:0.9.10.1`

simensaxe, 2024-03-01:
Thank you, will do so!

closed 2024-03-01 by simensaxe