autonomousvision/carla_garage #34: The question about plant

raozhongyu, 2024-03-29:
I notice that the input route and the output  checkpoint ground truth is same.  How it works?

Kait0, 2024-03-29:
The checkpoints predict the route centerline, they are the same. Since the route centerline is available in the HD-map there is no reason to predict it, one could also directly use it (e.g.: like in [map_agent](https://github.com/autonomousvision/carla_garage/blob/ce3df9525344133399a3275c1295758164a2fb4b/team_code/map_agent.py#L667C82-L667C100)).
We predict it anyway with the disentangled representation in this plant implementation. There is no particular reason for it. In some sense it makes sure that the architecture has no information bottleneck that prevents it from copying its input.

raozhongyu, 2024-04-01:
Thanks a lot. I get it.  Have you tried only predict the target speed prediction?

Kait0, 2024-04-03:
no

closed 2024-04-05 by raozhongyu

raozhongyu, 2024-04-06:
Hi, I have try to train the model on my own computer, but it seems have a poor results which only can achieve the 60 Ds. I don't find the pretrained models with plant model. Can you provide it?