autonomousvision/carla_garage #38: training routes for s

buaazeus, 2024-06-25:


Kait0, 2024-06-25:
S1 is control loss (rubble on the road). S3 is the pedestrian crossing scenario.
Both of them can be placed at basically any location, so it is probably not an issue that they occur at the same location.

buaazeus, 2024-06-26:
OK, Got it. 
The Scenario order are different from leaderboard.carla.
Thank you.

NUMBER_CLASS_TRANSLATION = {
    "Scenario1": ControlLoss,
    "Scenario2": FollowLeadingVehicle,
    "Scenario3": DynamicObjectCrossing,
    "Scenario4": VehicleTurningRoute,
    "Scenario5": OtherLeadingVehicle,
    "Scenario6": ManeuverOppositeDirection,
    "Scenario7": SignalJunctionCrossingRoute,
    "Scenario8": SignalJunctionCrossingRoute,
    "Scenario9": SignalJunctionCrossingRoute,
    "Scenario10": NoSignalJunctionCrossingRoute
}

https://leaderboard.carla.org/scenarios/

closed 2024-06-26 by buaazeus

Kait0, 2024-06-27:
The scenario description on the website have changed since this paper. This is the description of the leaderboard 2.0 scenarios.
I am not sure if the old leaderboard 1.0 scenario descriptions are still available.