autonomousvision/carla_garage #43: can't find file global_route_planner_dao

orangelee89, 2024-08-17:
[image]
when I run  leaderboard_evaluator_local.py got the error .

orangelee89, 2024-08-17:
and also I can't find the file in the repo

Kait0, 2024-08-17:
These files are in the CARLA folder (version 0.9.10.1, they have been changed in newer CARLA versions).
path/to/CARLA/PythonAPI/carla/agents/navigation
You need to add this folder to the python path as is done for example [here](https://github.com/autonomousvision/carla_garage/blob/1448bed63642bfe7c25e49e79e15713b20d104e6/leaderboard/scripts/local_evaluation.sh#L6).

closed 2024-08-19 by Kait0

orangelee89, 2024-08-20:
hi,thanks for your answering ,that's helpful.
but now I saw  "export TEAM_CONFIG=${WORK_DIR}/model_ckpt/transfuser"   in "local_evaluation.sh" ,I didn't see this folder in "carla_garage"

Kait0, 2024-08-20:
TEAM_CONFIG points to the folder containing the model you want to evaluate.
/model_ckpt/transfuser is the folder where I usually store the model weights and config file in.
You have to create it yourself and copy the model that you want to evaluate into that folder.

orangelee89, 2024-08-20:
Hi ,when I evaluate pre-train model , I have the error below :

========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent

**Could not set up the required agent:
> 'SensorAgent' object has no attribute 'config'**

Traceback (most recent call last):
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 294, in _load_and_run_scenario
    self.sensors = self.agent_instance.sensors()
  File "/home/carla/carla/carla_garage/team_code/sensor_agent.py", line 219, in sensors
    'x': self.config.camera_pos[0],
AttributeError: 'SensorAgent' object has no attribute 'config'
> Registering the route statistics
Traceback (most recent call last):
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 294, in _load_and_run_scenario
    self.sensors = self.agent_instance.sensors()
  File "/home/carla/carla/carla_garage/team_code/sensor_agent.py", line 219, in sensors
    'x': self.config.camera_pos[0],
AttributeError: 'SensorAgent' object has no attribute 'config'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 494, in main
    leaderboard_evaluator.run(arguments)
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 437, in run
    self._load_and_run_scenario(args, config)
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 325, in _load_and_run_scenario
    result = self._register_statistics(config, route_date_string, args.checkpoint, entry_status, crash_message)
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 249, in _register_statistics
    self.statistics_manager.save_record(current_stats_record, config.index, checkpoint)
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/utils/statistics_manager_local.py", line 259, in save_record
    save_dict(endpoint, data)
  File "/home/carla/carla/carla_garage/leaderboard/leaderboard/utils/checkpoint_tools.py", line 76, in save_dict
    with open(endpoint, 'w') as fd:
FileNotFoundError: [Errno 2] No such file or directory: '/home/carla/carla/carla_garage/results/transfuser_plus_plus_longest6.json'

Kait0, 2024-08-21:
Hm well this indicated that the [setup](https://github.com/autonomousvision/carla_garage/blob/1448bed63642bfe7c25e49e79e15713b20d104e6/team_code/sensor_agent.py#L69) function was not called properly or crashed. 
Perhaps something with the paths is wrong and it called the setup of the base class instead, or the setup function crashed due to not finding config.pickle.
You need to debug to see what is the problem.

orangelee89, 2024-08-23:
I see,  but in **leaderboard_evaluator_local.py** below, how I get the check point. I want to use pretrain model to evaluate .
    
parser.add_argument("--checkpoint", type=str, default='./simulation_results.json',
                        help="Path to checkpoint used for saving statistics and resuming")

Kait0, 2024-08-23:
--checkpoint should point to a json file where the results of the evaluation are stored.
To configure the agent you need to set:
--agent-config
It needs to point to a folder, and the folder should contain the model_0030.pth and config.pickle.