autonomousvision/carla_garage #44: PDM-Lite evaluation crashing

PanZWarzywniaka, 2024-08-22:
Hi there,

I tried to evaluate your PDM-Lite sitting on the Leaderboard_2 branch.

I tried to follow the setup correctly, downloaded CARLA 0.9.14 activated conda env and setup variables.

When running `start_expert_local.sh` CARLA starts, scenario and the agend loads up, but then at the first turn the process crashes due to this error.

I include the video recording of the bug: https://youtu.be/V-9GcmAwjqU
As well as script used to evaluate: https://pastebin.com/SdK6D5Ew
And full logs: https://pastebin.com/AFqnX2Hk

Highlight error message:
```
Stopping the route, the agent has crashed:

Traceback (most recent call last):
  File "/home/olek/Documents/dev/carla_garage/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 181, in _tick_scenario
    ego_action = self._agent_wrapper()
  File "/home/olek/Documents/dev/carla_garage/leaderboard/leaderboard/autoagents/agent_wrapper_local.py", line 147, in __call__
    return self._agent(self.sensor_list_names)
  File "/home/olek/Documents/dev/carla_garage/leaderboard/leaderboard/autoagents/autonomous_agent_local.py", line 114, in __call__
    control = self.run_step(input_data, timestamp, sensors)
  File "/home/olek/anaconda3/envs/pdm_lite/lib/python3.7/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/home/olek/Documents/dev/carla_garage/team_code/data_agent.py", line 324, in run_step
    tick_data = self.tick(input_data)
  File "/home/olek/Documents/dev/carla_garage/team_code/data_agent.py", line 275, in tick
    bounding_boxes = self.get_bounding_boxes(lidar=lidar_360)
  File "/home/olek/Documents/dev/carla_garage/team_code/data_agent.py", line 470, in get_bounding_boxes
    is_opposite = ego_wp.lane_id * wp.lane_id < 0
AttributeError: 'NoneType' object has no attribute 'lane_id'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator_local.py", line 391, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/home/olek/Documents/dev/carla_garage/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 158, in run_scenario
    self._tick_scenario()
  File "/home/olek/Documents/dev/carla_garage/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 189, in _tick_scenario
    raise AgentError(e)
leaderboard.autoagents.agent_wrapper_local.AgentError: 'NoneType' object has no attribute 'lane_id'
```

I would appreciate any help solving this problem.
Best regards :)

Jens21, 2024-08-23:
Hi,
thanks for reporting this. We've identified the issue - two lines were in the wrong order, causing the 'NoneType' error. We've fixed this bug and pushed the update to the leaderboard_2 branch.
Let us know if you need any further assistance.
Best, 
Jens

PanZWarzywniaka, 2024-08-23:
Hi Jens,

Thank you very much for the quick resolution.
Evaluations now work as expected.

Best
Olek Osikowicz

closed 2024-08-23 by PanZWarzywniaka