autonomousvision/carla_garage #47: Question about result reproduction on the Longest6 benchmark

YzMark2r, 2024-09-25:
Thank you for your work. We have encountered some difficulties while trying to reproduce Longest6 benchmark and would like to seek your advice. On a Tesla T4 server, we followed the README in the repository and used the weights you provided, resulting in scores of 54.2 DS, 81.986 RC, and 0.684 IS. We are concerned that differences in GPU versions may have led to performance discrepancies, or that we may have made some errors in our implementation.

Could you please advise us on how we can effectively reproduce your results? Should we consider re-collecting data and training the model on our servers to achieve better performance?

Thank you for your assistance!

Kait0, 2024-09-25:
You would need to provide some additional information for me to help you.

Which model files have you evaluated exactly (with what settings)?
Can you post some image of how the visualization looks like during the run?
Can you share the full results (result.json files or the .csv from the result parser)?

It is possible that there are some problems with the hardware. The RTX 3XXX series is for example known to produce buggy renderings. I am not sure about Tesla T4.
I would not expect the difference to be this large though, from rendering bugs, so recollecting the dataset would not be the first course of actions. 
More likely that there is some other problem like crashed routes in the results or something like that.

YzMark2r, 2024-09-26:
Thank you for your response. 

1. We used the tfpp_all_0 model along with the default configuration file and set UNCERTAINTY_THRESHOLD=0.33. Due to issues with the graphics driver, the command to start CARLA was ‘./CarlaUE4.sh -vulkan’, and we were unable to successfully launch it with ‘-opengl’. However, we also tried starting CARLA with the command ‘DISPLAY= ./CarlaUE4.sh -opengl’, and the test results were similar to those obtained with ./CarlaUE4.sh -vulkan.
2. The visualization images appear to be correct, as shown in the figure below.
3. We have uploaded the result files  as follows.
4. We followed the instructions in the README for reproduction, and aside from the CARLA startup command, there were no other differences.
5. During our experiments, we encountered several timeouts due to traffic congestion, although we adhered to the default setting of --timeout=600. Could this be the reason for the low scores, and how can we address this issue?

Thank you once again for your help!
[image]
[image]
[image]
[results.csv](https://github.com/user-attachments/files/17150861/results.csv)
[transfuser_plus_plus_longest6.json](https://github.com/user-attachments/files/17150864/transfuser_plus_plus_longest6.json)

Kait0, 2024-09-26:
Hm, I don't see something that is obviously wrong.
You vehicle collisions (0.83 to 1.02) are higher than usual as well as the blocked metric (0.06 to 0.19).
The other metrics look fine.
The blocked increase in particular can have a large impact on DS.

> During our experiments, we encountered several timeouts due to traffic congestion, although we adhered to the default setting of --timeout=600. Could this be the reason for the low scores, and how can we address this issue?

The --timeout=600 value is the timeout value for the code not the model (seems like it was not triggered).
Some timeouts during evaluation are normal.

You only ran 1 evaluation seed. We usually run 3 evaluation runs and average the results.
In your case I would first run 3 repetitions, to make sure you are not wasting time due to an unlucky seed.

Is it possible for you to run the evaluation with -opengl? Even though both opengl and vulkan might look good to a human eye they can look different to a neural network (network was trained with opengl data).
If the problem is related to the display you could run CARLA in offscreen mode.
SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 {carla_root}/CarlaUE4.sh  -opengl

YzMark2r, 2024-09-27:
Thank you for your response and assistance. We will take a few days to conduct experiments based on your suggested methods, and we will update you on their effectiveness afterward.

Kait0, 2024-10-07:
Guess it is solved. Let me know if you still have problems.

closed 2024-10-07 by Kait0