autonomousvision/carla_garage #48: Some problems when generating the dataset

hollowknight2167, 2024-10-08:
Hello, thank you for your incredible work, I am working on rebuild a new dataset base on your method: Change the `DATAGEN` environment variable to 1 and run the `leaderboard_evaluator_local.py` to collect data. And I'd like to ask you for some advice.

However, the result shows the data_agent still get crashed during the simulation. I wonder whether this situation is common or not. If not, what can I do to avoid this bug? By the way, given that currently I can only find the collision after simulation, I'd like to know if the code could give me a feedback in time, so that I can split the dataset and use the valid part.

The route file I used is `routes_training.xml`
The agent I used is `data_agent.py`

[image]

Kait0, 2024-10-08:
I assume you mean vehicle crashes (and not software crashes here).
Yes the data_agent is not a perfect driver is also gets some collisions (as you can see in Table 6 and 7 under the Veh metric).
We skip those episodes during training by only considering data with 100 DS, see [here](https://github.com/autonomousvision/carla_garage/blob/1448bed63642bfe7c25e49e79e15713b20d104e6/team_code/data.py#L92).

There is no way to stop data collection on crashed routes in the released code. 
You could implement it yourself if you want to, should not be too hard.

You can check the amount of routes where the vehicle crashed in the released dataset by running a training.
The value gets printed [here](https://github.com/autonomousvision/carla_garage/blob/1448bed63642bfe7c25e49e79e15713b20d104e6/team_code/data.py#L229C1-L233C1).

hollowknight2167, 2024-10-09:
Thank you for your advice! I will take your advice into consideration and proceed with another experiments to collect data.

closed 2024-10-09 by Kait0