autonomousvision/carla_garage #50: Question About Two-Stage Training Setup

renxinxing123, 2024-11-17:
Hello,

Thank you for sharing your excellent work on Transfuser++. I really appreciate the clarity and depth of your paper—it’s been very inspiring and helpful for my research.

I have a question regarding the two-stage training mentioned in your paper. For the second stage, it’s stated that the learning rate is set to 0.0003 and reduced to 0.00003 at the 30th epoch, with a total of 31 epochs. I’d like to confirm:

1.Are the learning rate and epoch settings the same for the first stage?
2.For the first stage, is the only required parameter change to set use_controller_input_prediction to 0?
Thank you in advance for your time and assistance!

Best regards,
Xinxing

Kait0, 2024-11-18:
1. yes
2. yes, --use_controller_input_prediction 0 --use_wp_gru 0 both need to be 0 for the first stage. In the second stage you set the load_file argument to the last model of the first stage and --continue_epoch 0.

renxinxing123, 2024-11-18:
> 1. yes
> 2. yes, --use_controller_input_prediction 0 --use_wp_gru 0 both need to be 0 for the first stage. In the second stage you set the load_file argument to the last model of the first stage and --continue_epoch 0.

Thank you very much for your prompt reply!! Will try it!

closed 2024-11-18 by renxinxing123