autonomousvision/carla_garage #51: RuntimeError: Trying to resize storage that is not resizable during training

renxinxing123, 2024-11-19:
Hi, I’m trying to train the first stage of Transfuser++ using 4 NVIDIA 4090 GPUs. The command I used is:

`CUDA_VISIBLE_DEVICES=0,1,2,3 OMP_NUM_THREADS=36 OPENBLAS_NUM_THREADS=1 torchrun --nnodes=1 --nproc_per_node=4 --max_restarts=0 --rdzv_id=1234576890 --rdzv_backend=c10d train.py --logdir /mnt/carla_garage/log --root_dir /mnt/carla_garage/data --id tf++_1st_stage --cpu_cores 36
`

Due to memory limitations (each GPU uses ~22GB of its 24GB memory during training), I set the batch size to 12 per GPU. The first epoch completes without issues, but during the second epoch, I encounter the following error:

`Root Cause (first observed failure):
[0]:
  time      : 2024-11-19_22:16:55
  host      : ZA3M9O
  rank      : 2 (local_rank: 2)
  exitcode  : 1 (pid: 251760)
  error_file: /tmp/torchelastic_w2sb6ux9/1234576890_14q1mjhw/attempt_0/2/error.json
  traceback : Traceback (most recent call last):
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 345, in wrapper
      return f(*args, **kwargs)
    File "train.py", line 624, in main
      trainer.train()
    File "train.py", line 883, in train
      for i, data in enumerate(tqdm(self.dataloader_train, disable=self.rank != 0)):
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/tqdm/std.py", line 1183, in __iter__
      for obj in iterable:
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 681, in __next__
      data = self._next_data()
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1376, in _next_data
      return self._process_data(data)
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1402, in _process_data
      data.reraise()
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/_utils.py", line 461, in reraise
      raise exception
  RuntimeError: Caught RuntimeError in DataLoader worker process 7.
  Original Traceback (most recent call last):
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 302, in _worker_loop
      data = fetcher.fetch(index)
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 52, in fetch
      return self.collate_fn(data)
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/_utils/collate.py", line 160, in default_collate
      return elem_type({key: default_collate([d[key] for d in batch]) for key in elem})
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/_utils/collate.py", line 160, in <dictcomp>
      return elem_type({key: default_collate([d[key] for d in batch]) for key in elem})
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/_utils/collate.py", line 149, in default_collate
      return default_collate([torch.as_tensor(b) for b in batch])
    File "/root/miniconda3/envs/garage/lib/python3.7/site-packages/torch/utils/data/_utils/collate.py", line 140, in default_collate
      out = elem.new(storage).resize_(len(batch), *list(elem.size()))
  RuntimeError: Trying to resize storage that is not resizable`

Steps I have taken to debug this issue:

1.  Re-downloaded the dataset: I suspected the issue might be related to the dataset, but re-downloading it didn’t resolve the error.
2.  Reduced batch size to 6: I suspected a specific batch might be exceeding the GPU memory limit, so I further reduced the batch size to 6, but the same issue occurred during the second epoch.

It seems the error might be related to a `DataLoader` operation or memory allocation during the second epoch.

Could you please advise on what might be causing this error and how to resolve it? Any insights or recommendations would be greatly appreciated!

Thank you!

renxinxing123, 2024-11-20:
And plus，this is the machine I am using 'GPU
4 × NVIDIA GeForce RTX 4090
CPU
40 × Intel(R) Xeon(R) Gold 6338 CPU @ 2.00GHz'

Kait0, 2024-11-20:
Maybe [this](https://github.com/autonomousvision/carla_garage/issues/28)?

renxinxing123, 2024-11-20:
> Maybe [this](https://github.com/autonomousvision/carla_garage/issues/28)?

Seems that the problem solved! It is smoothly running at the 3rd epoch. Thank you very much! A big heart for you ❤️ ❤️ ❤️ ❤️ ❤️!!

closed 2024-11-20 by renxinxing123