autonomousvision/carla_garage #52: Data Difference

wyddmw, 2024-12-25:
Hi, thanks for this awesome dataset for Leaderboard 2.0! As mentioned in the README, the data provided via the download script is slightly older than the latest one used in the paper. I would like to know whether this older version will affect the final performance after training. And if that is the case, how much does the result differ?

Kait0, 2024-12-25:
It did not seem to affect the performance on the benchmarks I tested it (bench2drive, Town 13). I could not test it on the leaderboard since it is closed.

These two commits ([1](https://github.com/autonomousvision/carla_garage/commit/6f00e946c23b29dc83b3bc64cc320c5bdac0761b) [2](https://github.com/autonomousvision/carla_garage/commit/af35ce5f240c6d9820033873263be669d656a340)) are not included.

wyddmw, 2024-12-26:
> It did not seem to affect the performance on the benchmarks I tested it (bench2drive, Town 13). I could not test it on the leaderboard since it is closed.
>
> These two commits [...]

Thanks for your kind reply! I will close this issue.

closed 2024-12-26 by wyddmw