autonomousvision/carla_garage #53: Model is not behaving as expected

Kaiser-75, 2025-01-30:
I am using leaderboard 2.0
Let's say there is a leading vehicle in front of the agent, and the agent decelerates and stops. The agent should change lanes and proceed to the destination.

The same should happen when there is an obstruction in its path; the agent should change lanes to reach its destination. However, it gets stuck and does not change lanes to overcome the obstacles.

Kait0, 2025-01-30:
So the agent in the leaderboard 2.0 repo is trained to change lanes for particular scenarios.
1st possibility is that the model does not have perfect performance and fails at these scenarios sometimes.
So maybe there are different variations of your scenario that the model can solve.

To make sure your setup is correct, I would advise you first test with some of the existing scenarios to see that the model behaves correctly there:
perhaps:
Maybe you can look at some of the [individual Bench2Drive routes](https://github.com/autonomousvision/carla_garage/tree/leaderboard_2/leaderboard/data/bench2drive_split) for a quick example:

The second is that the model might not generalize to new types of obstacles. I think cars in the CARLA leaderboard 2.0 always start driving again eventually and don't stop indefinitely in the middle of the road.
If you want to solve new scenarios, then maybe you need to include demonstrations in the training data of those scenarios.
Currently, CARLA papers make no claims about generalizing to new scenario types (only different instances of similar scenarios). All scenario types that the model solves have also been observed during training.

Kaiser-75, 2025-01-30:
Got you.
I used this specific one:
<routes>
   <route id="1" town="Town02">
      <waypoints>
         <position x="7.6" y="109.9" z="0.0"/>	
         <position x="121.0" y="108.0" z="0.0"/>
      </waypoints>
      <scenarios>
         <scenario name="ConstructionObstacle_1" type="ConstructionObstacle">
            <trigger_point x="20" y="109.9" z="0.0" yaw="0.0"/>
            <distance value="13"/>
            <direction value="right"/>
            <speed value="0"/>
         </scenario>
      </scenarios>
   </route>
</routes>

Kait0, 2025-02-03:
I do not observe the problem you describe. For me when I run the all towns model on your route the model does overtake the construction site.

[image]

So there is probably some problem with your setup.
Can you maybe set the environment variable
SAVE_PATH=/path and DEBUG_CHALLENGE=1
to see some visualizations of what is going on?
Also what are the variables that are printed at the beginning of the route?

Note that the scenario definition is also potentially impossible to solve. The construction site gets spawned very late while the car is driving at high speed.

[image]

The model reacts immediately but it is already too late to avoid the collision with the construction sign.
[image]

Kaiser-75, 2025-02-05:
okay, I will get back to you soon.

Kaiser-75, 2025-02-05:
Where is "all town model"?

I will try as per your suggestion.

Kait0, 2025-02-05:
I was refering to the folder where the model is stored from the [provided models](https://github.com/autonomousvision/carla_garage?tab=readme-ov-file#pre-trained-models).

closed 2025-02-11 by Kaiser-75