autonomousvision/carla_garage #56: About downloading the dataset generated by PDM-Lite

liyingyanUCAS, 2025-02-18:
Hi, thank you for your excellent work. Regarding the [carla_garage](https://github.com/autonomousvision/carla_garage/tree/leaderboard_2)/[tools](https://github.com/autonomousvision/carla_garage/tree/leaderboard_2/tools)/download_data.sh, would it be possible to provide a download link via Hugging Face? This would help users in China access the data through accelerated mirrors, allowing for faster download speeds.

liyingyanUCAS, 2025-02-19:
Thank you for your quick reply. At the moment, the primary bottleneck for my downloads is the slow VPN speed. Using a site like hf-mirror could help bypass the limitations imposed by the VPN. I’m sorry to hear that Hugging Face may introduce high fees in the future. If hosting on Hugging Face is not feasible, I completely understand and will explore other options.

Kait0, 2025-02-21:
There is a PDM-Lite dataset on Hugging Face, you can find it [here](https://huggingface.co/datasets/autonomousvision/PDM_Lite_Carla_LB2).
It is an older version of the dataset released in this repo, but should give similar performance.
The folder structure is not the same, so you might need to adapt the data loader if you want to load it with the carla_garage code.
Also the HD-MAP in Town 13 is lower resolution, so you might need to upscale it.

I am in principle fine to host a mirror on Hugging Face as long as it stays free.
Will let you know when I get around to upload it.

There is of course also the option for you to purchase access to a high speed VPN, but that might cost some extra money.

closed 2025-02-21 by liyingyanUCAS