autonomousvision/carla_garage #57: Semantic Segmentation Model - During inference

Alobaali12, 2025-02-25:
Hi, 

I hope my message finds you well. 

Would you please provide some guidance on how I would use a pre-training carla semantic segmentation model to generate features from the input image and then feed that to my model to do the predictions, just so that my model doesn't only rely on raw pixels of the input image but also has some feature understanding. 

Is there any open source pre-trained models I can use for this ? 

Can I use the Conv decoder branch from Transfuser ? (is it easy to just use / extract that part of transfuser) (note my model uses images only !)? 

Similar to this:

[image]

[image]

Thank you.

Kait0, 2025-02-25:
Not sure if there is a pure image based semantic segmentation model. You could train a AIM with semantic segmentation head or use [Latent TransFuser](https://github.com/autonomousvision/transfuser). Note that these auxiliary tasks and decoders are not optimized for high quality semantic segmentation (they don't have skip connections for example).

If you reproduce an actual semantic segmentation architecture in CARLA you might get higher quality results.

Alobaali12, 2025-02-25:
Thanks Kait0. 

If I may ask, is there a reason why the semantic segmentation image produces by the PDM_LITE Dataset looks like this :

[image]

As in why is the feature map not like the Carla generated semantic segmentation, colored (with multiple classes).

closed 2025-02-27 by Kait0