autonomousvision/carla_garage #58: pretrained model & huggingface problem

chan-yuu, 2025-03-08:
Hello, I can't get the pre-trained model through huggingface, can you provide it?
```

Traceback (most recent call last):
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/huggingface_hub/file_download.py", line 1376, in _get_metadata_or_catch_error
    metadata = get_hf_file_metadata(
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/huggingface_hub/utils/_validators.py", line 114, in _inner_fn
    return fn(*args, **kwargs)
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/huggingface_hub/file_download.py", line 1296, in get_hf_file_metadata
    r = _request_wrapper(
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/huggingface_hub/file_download.py", line 280, in _request_wrapper
    response = _request_wrapper(
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/huggingface_hub/file_download.py", line 303, in _request_wrapper
    response = get_session().request(method=method, url=url, **params)
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/huggingface_hub/utils/_http.py", line 96, in send
    return super().send(request, *args, **kwargs)
  File "/home/cyun/anaconda3/envs/garage_2/lib/python3.10/site-packages/requests/adapters.py", line 507, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: (MaxRetryError("HTTPSConnectionPool(host='huggingface.co', port=443): Max retries exceeded with url: /timm/regnety_032.ra_in1k/resolve/main/pytorch_model.bin (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7f7b61c14640>, 'Connection to huggingface.co timed out. (connect timeout=10)'))"), '(Request ID: 133fc6aa-3510-47b9-9b6d-b589899cdec8)')
```

```
class TransfuserBackbone(nn.Module):
  """
    Multi-scale Fusion Transformer for image + LiDAR feature fusion
    """

  def __init__(self, config):
    super().__init__()
    self.config = config

    **self.image_encoder = timm.create_model(config.image_architecture, pretrained=True, features_only=True)**

    self.lidar_video = False
    if config.lidar_architecture in ('video_resnet18', 'video_swin_tiny'):
      self.lidar_video = True
```

chan-yuu, 2025-03-08:
I downloaded it https://huggingface.co/timm/regnety_032.ra_in1k/resolve/main/pytorch_model.bin
transfuser.py:
```
    # self.image_encoder = timm.create_model(config.image_architecture, pretrained=True, features_only=True)

    self.image_encoder = timm.create_model(
        config.image_architecture, 
        pretrained=False,  # ban
        features_only=True
    )
    # self.image_encoder.load_state_dict(torch.load("/home/cyun/Downloads/apps/CARLA_0.9.15/carla_garage/pytorch_model.bin"))
    self.image_encoder.load_state_dict(
        torch.load("/home/cyun/Downloads/apps/CARLA_0.9.15/carla_garage/pytorch_model.bin"),
        strict=False  # fix
    )
```

I want to see what the visual pygame interface should do

[image]

closed 2025-03-13 by Kait0

chan-yuu, 2025-03-17:
Thank you for your reply.
I'm sorry, I have a problem with my presentation. I was just wondering if there was an interface for Pygame visualization because I wanted to see the perspective of the front of the vehicle.

Kait0, 2025-03-17:
You can check the visualization function see [here](https://github.com/autonomousvision/carla_garage?tab=readme-ov-file#local-evaluation-and-debugging) and [here](https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/team_code/model.py#L675).
They work with opencv not pygame.