autonomousvision/carla_garage #61: Kinematic Bicycle model Parameters

Alobaali12, 2025-03-20:
[image]

Hi, 

I hope my message finds you well. 

I would like to ask about the kinematic bicycle model parameters in the config.py file, how come  "self.front_wheel_base" equals -0.090769015 ??

First, it should be a scalar not a vector and also why is it small ? 

[image]

Btw, do you know what vehicle the carla leaderboard 2.0 uses ? I assumed it's a tesla model 3, is that correct ? 

I ran some code in carla to get the centre of the vehicle and then calculated the relative distance of each wheel base (axle) from the COG, and I got the numbers shown in the above image (bottom).

May you please clarify ? 

Thank you.

Kait0, 2025-03-20:
The CARLA leaderboard uses an vehicle.lincoln.mkz_2017

These numbers were tuned in the [World on Rails](https://arxiv.org/abs/2105.00636) paper we copied them.
They used a dataset and gradient descent to fit them, that is why they are so weird.

The numbers work reasonably well for our purpose that is why we kept them for some problems like the filter. But they are likely not optimal particularly because they were tuned with a different simulator version.

Alobaali12, 2025-03-20:
Thanks Kait0. 

What do you mean by tuned ? I'm asking because I'm designing a nonlinear MPC controller to execute the waypoint predicitons from the model rather than PID. 

Carla has a vehicle physics control "carla.VehiclePhysicsControl" https://carla.readthedocs.io/en/0.9.9/python_api/#carla.VehiclePhysicsControl

[image]

Inside that there is a wheel physics control, which has a position attribute for wheel object.

[image]

Therefore a direct calculation can be performed, to find lr and lf, so why do need any esitmations ? or optimization ? 

I just wanna make sure I got the correct numbers for my vehicle model. 

Thank you.

Kait0, 2025-03-20:
they just collected a dataset of trajectories and fit the parameters with gradient descent.
I don't know if you need to do that but it works.

You might want to re-tune/set them yourself if you need high accuracy.

Alobaali12, 2025-03-20:
okay, thank you.

closed 2025-03-23 by Kait0