autonomousvision/carla_garage #63: Scenario Setup Errors in PDM Lite Evaluation

pirotyyy, 2025-04-03:
I'm encountering lane detection errors when running PDM Lite on the longest6_split routes. Has anyone experienced similar issues or have insights into what might be causing them?

## Issue
When running PDM Lite agent on the routes in `leaderboard/data/longest6_split/`, several scenarios are being skipped due to lane detection errors during setup.

**Specific error messages:**
```
[1m========= Preparing RouteScenario_10 (repetition 0) =========[0m
[1m> Loading the world[0m
[93mSkipping scenario 'OppositeVehicleRunningRedLight_2' due to setup error: Couldn't find a lane for the given direction
[0m[1m> Setting up the agent[0m
[1m> Running the route[0m
[93mSkipping scenario 'OppositeVehicleRunningRedLight_1' due to setup error: Couldn't find a lane for the given direction
[0m[93mSkipping scenario 'SignalizedJunctionRightTurn_2' due to setup error: Trying to find a lane in the left direction but none was found
```

## Steps to Reproduce
Use `evaluate_routes_slurm_pdm_lite.py` with slurm shell script similar to `run_evaluation_slurm_plant.sh` etc.
Or use data generation code (`collect_dataset_slurm.py`) with route_folder set to `longest6_split`

## Environment
- CARLA: 0.9.15 (installed by `setup_carla.sh`)
- Python: 3.10.15

Kait0, 2025-04-03:
These are normal.
The scenarios were autoconverted with the official scenario converter from the leaderboard 1.0 definition to the leaderboard 2.0 definition. 
This did not work for all scenarios because the way the scenarios work has changed.
So you see some scenarios that can not be set up.
(Should only be some scenarios not all of them).

pirotyyy, 2025-04-03:
Thank you for the explanation! I now understand that these lane detection errors are expected due to issues in the conversion process from Leaderboard 1.0 to 2.0 definitions.

I just want to make sure — is it correct to assume that these skipped scenarios do not significantly affect the overall evaluation results on the `longest6` routes?

Also, I noticed another warning during the same run:
```
carla_garage/team_code/kinematic_bicycle_model.py:47: RuntimeWarning: invalid value encountered in cast steers, throttles, brakes = actions[:, 0], actions[:, 1], actions[:, 2].astype(np.uint8)
```

Do you happen to know what might be causing this warning?  
It seems to be related to casting values to `uint8` in the kinematic bicycle model, but I'm not entirely sure.

Kait0, 2025-04-03:
yes, all longest6 v2 evaluations should have the same warnings, so results are comparable.
You should not compare longest6 v2 with longest6 v1 numbers (Different metric, scenario definitions simulator version etc.)

I also saw these warnings. Sometimes the CARLA brake value returns nan for some cars. These just gets mapped to 0 during the uint8 conversion here and should not affect performance.

pirotyyy, 2025-04-03:
Thank you so much for the clarification!  I really appreciate your help.

closed 2025-04-03 by pirotyyy

Kait0, 2025-04-03:
welcome. PDM-Lite should get something like 73 DS on longest6 v2