autonomousvision/carla_garage #65: Missing results on the Bench2Drive benchmark

yixiao1, 2025-04-15:
Hi there,

Thanks a lot for the impressive work!

I found that in Table 4, the main results table on Bench2Drive benchmark, there are two results of metrics Efficiency and Comfortness missing. Any reason for not showing them? Or if you put them into the supplementary, could you please provide the material as I couldn't find it anywhere?

Cheers,
Yi

Kait0, 2025-04-15:
I think these metrics just need extra code since they were added in a later version of bench2Drive, and we didn't run the code.
You can find the numbers for PDM [here](https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/Bench2Drive/pdm_lite_b2d_traj/merged_ability.json#L8).
I don't have the numbers for TF++, but I think the repository should contain everything you need to evaluate them yourself.
(Eval provided checkpoint on Bench2Drive and run the metric calculation script).

yixiao1, 2025-04-16:
Awesome! Thanks a lot for your reply.

closed 2025-04-16 by yixiao1