autonomousvision/carla_garage #66: Visualisation issues

bujikuibu, 2025-04-16:
Very nice work, I would like to ask how the visualisation of auxiliary tasks is implemented here? And also this question I didn't really see, what about after commenting this out?https://github.com/autonomousvision/carla_garage/issues/14#issuecomment-1770826904

Kait0, 2025-04-16:
These were made with image editing software, Inkscape for example.

The black dots in the image are a result of setting --debug > 0 on the leaderboard_evaluator.py.
You don't need to do that to get the other visualizations.

When you comment this the visualized map should be ground truth instead of what the model predicts.

bujikuibu, 2025-04-17:
Thank you for your reply.

> These were made with image editing software, Inkscape for example.
> Does the depth image and scene segmentation image here provide a visualisation of the code? I saw before that your TransFuser works to show both images, but I don't see it in the code TransFuser++.
> The black dots in the image are a result of setting --debug > 0 on the leaderboard_evaluator.py. You don't need to do that to get the other visualizations.
> Here I do have debug set to 1. Is it possible to remove this black path point?
> When you comment this the visualized map should be ground truth instead of what the model predicts.
>IIf I want to view this real visualisation of the map, can you tell me the exact details of how to do it? After I uncomment this [link](https://github.com/autonomousvision/carla_garage/blob/8bb143807ee9df6c1524fe72043ea687ad2ec2aa/team_code/model.py# L707), do I still need to comment the bev scene segmentation code?

Thanks again for your reply, thanks!

Kait0, 2025-04-23:
You need to set --debug=0
No the depth and semantic visualization is not included.