autonomousvision/carla_garage #69: Agents and their names

MasoudJTehrani, 2025-05-14:
Good morning,
I hope you're doing well. 
In the pre-trained weights files, you have 

- lav/tfpp_02_05_withheld
- lav/aim_02_05_withheld
- leaderboard/tfpp_wp_all
- longest6/tfpp_all

I wanted to know the name of each agent. For example, which one is transfuser++, what is the agent of longest or lav/aim or lav/tfpp, or which 
one is interfuser.

MasoudJTehrani, 2025-05-14:
Thank you very much. so I guess longest6 was the best-performing one yes?

Kait0, 2025-05-14:
Yes, the longest6 model is the best-performing one.

Well, the difference is the training seed. 
Each model was trained with a different random seed. 
The last number is an index to distinguish them.
Otherwise, they are the same.

The Plant model is a reproduction of the privileged [Plant](https://arxiv.org/abs/2210.14222) model, though we have not used it in our paper.
See [here](https://github.com/autonomousvision/carla_garage/blob/leaderboard_1/docs/additional_features.md).