autonomousvision/carla_garage #70: visualization issue

TensorsSun, 2025-05-14:
Hello, I would like to ask whether the visualization result will be generated after each scenario is tested or after all scenarios are tested.

Kait0, 2025-05-14:
If you [turn on visualization](https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/team_code/model.py#L889) images are stored at every simulator step see [here](https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/team_code/model.py#L889).

TensorsSun, 2025-05-14:
That's weird, I didn't see any visualization generated after testing a scene. I have to make sure this has nothing to do with the Carla simulator, right? Because I used ./CarlaUE4.sh -RenderOffScreen.

TensorsSun, 2025-05-15:
Another problem occurred. When I ran `./CarlaUE4.sh -RenderOffScreen` and then ran `bash Bench2Drive/leaderboard/scripts/run_evaluation_tf++.sh`, the evaluation program would be interrupted directly without any error message. However, CarlaUE4 continued to run. The evaluation program reacted as if it was directly evaluated without starting CarlaUE4. The following is the output of the log file:

```
CUDA_VISIBLE_DEVICES=0 python /mnt/hwdata/xiaolong/carla_garage/Bench2Drive/leaderboard/leaderboard/leaderboard_evaluator.py --routes=/mnt/hwdata/xiaolong/carla_garage/Bench2Drive/leaderboard/data/bench2drive220_0_tfpp_traj.xml --repetitions=1 --track=SENSORS --checkpoint=/mnt/hwdata/xiaolong/carla_garage/Bench2Drive/tfpp_b2d_traj/eval_bench2drive220_0.json --agent=/mnt/hwdata/xiaolong/carla_garage/team_code/sensor_agent.py --agent-config=/mnt/hwdata/xiaolong/carla_garage/results/tfpp_010_1 --debug=1 --record= --resume=True --port=30000 --traffic-manager-port=50000 --gpu-rank=0
/mnt/hwdata/xiaolong/carla_garage/carla/CarlaUE4.sh -RenderOffScreen -nosound -carla-rpc-port=30000 -graphicsadapter=0 None
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
load_world success , attempts=0
traffic_manager init success, try_time=0
cost time=4.5299530029296875e-06
[1m> Registering the global statistics[0m
```
Have you ever encountered a similar situation before?

Kait0, 2025-05-16:
when you render  -RenderOffScreen the spectator camera is turned off but the visualization should still be generated. You can debug to check why the code is not executed.

Bench2Drive starts the CARLA server by [itself](https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/Bench2Drive/leaderboard/leaderboard/leaderboard_evaluator.py#L210).

closed 2025-05-18 by Kait0