autonomousvision/carla_garage #71: Clarification on stopping distance at red lights and `idm_red_light_minimum_distance`

pirotyyy, 2025-05-18:
Hi, 

I'm using PDM-lite for data generation and have a couple of questions regarding the ego vehicle's stopping behavior at red traffic lights.

1.  I've observed that the ego vehicle often seems to stop **considerably before the actual stop line** when encountering a red light. I've attached an image illustrating this.
[image]

2.  The `idm_red_light_minimum_distance` is set to `6.0` meters in the `GlobalConfig`. Could you please share the reasoning or considerations behind choosing this specific 6.0m value? For example, was it for specific safety margins, pedestrian clearance, or other simulation/control objectives?

Best regards,
Hiroki

pirotyyy, 2025-05-18:
Thanks for creating this agent!

Following up on my initial questions, I have one more related point I'd like to ask about, if possible.

I'm considering whether adjusting the `idm_red_light_minimum_distance` to a shorter value  might help the vehicle stop closer to the actual stop line, which seems more aligned with typical real-world driving behavior in many cases.

I was wondering if you have, by any chance, experimented with shorter `idm_red_light_minimum_distance` values specifically for red light stopping scenarios? If so, were there any observed adverse effects that I should be aware of before attempting such an adjustment?

Any insights or past experiences you could share on this would be very helpful.

Thanks again!

Kait0, 2025-05-18:
> I've observed that the ego vehicle often seems to stop considerably before the actual stop line when encountering a red light. I've attached an image illustrating this.

This is so that EU style traffic lights (e.g. Town01) which are on the current side of the street can still be seen in the camera.
If you stop at the actual stop line then there are cases where the traffic light is not visible.

You could implement different logic for EU and US style traffic lights but we have used a joined one.

pirotyyy, 2025-05-19:
Hi Kait0,

Thanks a lot for the quick and clear answer.  
I understand the background now—really appreciate your help!

closed 2025-05-19 by pirotyyy