autonomousvision/carla_garage #78: Waypoint usage confusion.

wyddmw, 2025-07-21:
Hi Bernhard, thanks for your great work. When I run the evaluation script, I find the setting inconsistent as described in the paper. The pretrained model sets the parameter use_wp_gru as False, which is also the case in the training script.
https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/team_code/shell_train.sh#L17

With this setting activated, the gru-based trajectory planning module is skipped during the inference. May I ask if there is something I missed, or is this a new setting? Besides, the output from the model is a so-called checkpoint. I am confused about the definition of this checkpoint, and what is the difference between a checkpoint and a waypoint. According to the provided visualization script, the checkpoint is obtained from the route, which seems to be a point that the ego vehicle will reach in the future. And the waypoint is a local planned trajectory provided by the PDM_Lite? Please correct me if there is anything wrong! 

If I want to use the implementation of waypoint-based planning for TFPP, what should I modify to launch the evaluation?

Thanks in advance for your patience and feedback!

wyddmw, 2025-07-23:
> The default of TF++ uses the disentangled representation, which predicts the path (checkpoints) and classifies the target speed. use_wp_gru is for the entangled waypoint representation (TF++ WP from appendix).
> 
> Checkpoints are equally spaced apart and do not encode velocities. Waypoints predict the position of the car in the future, so they include velocities.
> 
> If you want to use TF++ WP (waypoint) you need to use the appropritate model weights and configure the eval script as described in the readme.

Hi Bernhard, I have already found the difference between these two definitions in the original TFPP paper. As demonstrated in the table3, leveraging the representation of path will efficiently improve the steering, which is indicated by the lower static obstacle collisions. May I ask what the fixed sampling interval is between each waypoint in the PDM_Lite dataset? 

[image]

Kait0, 2025-07-24:
1 meter distance between the path points if you use the [smooth_route feature](https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/team_code/config.py#L521).
No fixed distance if you do not.

Kait0, 2025-07-24:
TF predicts 2 seconds into the future. In this dataset this is [8 points](https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/team_code/config.py#L378).
https://github.com/autonomousvision/carla_garage/blob/d5e94cad950b52816fe81b858124e106c3ebef5c/team_code/config.py#L378