autonomousvision/carla_garage #79: Questions about collision and overtake behaviors

Yi-fe, 2025-07-23:
I appreciate your team’s sharing of your work. Recently, I have been trying to reproduce it on CARLA and have encountered a few issues I’d like to ask about. 
After setting up the traffic environment in Town 04 and deploying the model, I observed a peculiar phenomenon. In my case, background vehicles (BVs) spawn randomly around the model-controlled ego vehicle. When a BV spawns ahead of the ego in the same lane and its spawn point is within 15 m of the ego’s current position, the ego seems to ignore it entirely and simply collides with the BV. However, if the ego has just started moving or the BV is spawned more than 15 m away, the ego reacts properly and brakes in time. I wonder whether this is the expected behavior or if there is a configuration issue. I noticed a comment in sensor_agent.py: “Classification networks are known to be overconfident which leads to them braking a bit too late in our case.” I set the corresponding parameter to 1 to reduce the ego’s speed, but the issue persists. 
Additionally, when the ego encounters a slower-moving vehicle ahead, it merely follows it without attempting to overtake. Is this determined by the model’s control policy?

[image]

Kait0, 2025-07-24:
TF++ is trained not to overtake other vehicles unless in scenarios where they are static.

Not sure about the other problem. Have not observed this.
TF++ is a single-frame model, so temporal distance should not affect this.
The CARLA leaderboard scenarios sometimes spawn vehicles and then despawn them when setting up or finishing scenarios.

Yi-fe, 2025-07-28:
I understand. Thank you for your explanation. I’ll give it another try.

closed 2025-08-02 by Kait0