autonomousvision/carla_garage #90: Evaluation encountered issues

Wuwenjie-hive, 2025-09-02:
Hello, I encountered the following issues during evaluation: the agent kept crashing or couldn't be set up, and it consistently displayed a memory overflow error. Relevant information is as follows:

[image]I hope you can help me with this. Thank you.

Kait0, 2025-09-02:
You would need to post the actual error message you got on the console, but I suppose you just have too little RAM or GPU RAM if there is nothing wrong with the setup.

Wuwenjie-hive, 2025-09-02:
Alright, I'm currently running two tasks in parallel, but they keep failing. My classmate ran his evaluation on the same device without encountering this issue, so I'm quite puzzled. The terminal displays the following:

Wuwenjie-hive, 2025-09-03:
Thank you for your previous assistance. Now that the evaluation is complete, I'd like to ask how to handle this situation where the route completion score did not reach 100%. The specific details are shown in the image below:[image]

Kait0, 2025-09-09:
Blocked can just happen due to suboptimal model performance, this is usually not a bug.