autonomousvision/carla_garage #91: pretrained model

saibo2077, 2025-09-09:
Sorry to bother you, but do you still have the saved files for the separately trained perception model? I am referring to the model trained during the perception stage in the two-stage training. Additionally, I noticed that in the args.txt file of your pre-trained model, there is a parameter: use_point_pillars. Have you tried replacing the BEV histogram with PointPillars? Would this yield better performance?

Kait0, 2025-09-09:
Which model do you mean? Leaderboard 1.0 or 2.0?

We did not see a performance improvement with point pillars.
See [Table 9](https://www.cvlibs.net/publications/Chitta2022PAMI.pdf) in the TransFuser paper.

saibo2077, 2025-09-10:
Leaderboard 1.0。I would like to test the performance differences between MLP and Transformer decoders, but the pre-trained model you provided seems to be trained using a two-stage joint training setup. Could you kindly let me know if you have the pre-trained models saved for the individual perception stages? Thank you very much for your response.

saibo2077, 2025-09-10:
Leaderboard 1.0。I would like to test the performance differences between the MLP and Transformer decoders, but the pre-trained model you provided seems to be trained using a two-stage joint training setup. Thank you very much for your response.